makerbase-mks/MKS-Robin-Nano-V1.X #4: BED LEVELING (MESH) PROBLEM

pepe1125, 2020-01-11:
Hello,

I have an Two Trees Sapphire Plus 3D Printer, and I configured this firmware to Auto/Manual MESH bed leveling, and the firmware not corrected the bed level when I print. I used Two Tree Auto Level sensor, and manual mesh leveling. Is there a bug in the firmware? 

My Config:

#===================Advanced  Settings=======================================
>cfg_insert_det_module			0			#Assemble power outage module (1:mks 220det;0:mks pwc)

>cfg_have_ups_device			0			# UPS(1:yes;0:no)

>cfg_print_over_auto_close		1			# auto-off when print finishes(1:yesŁ»0:no)

>PRINT_FINISHED_COUNT 			180			#After the shutdown delay time (unit: second)

>cfg_filament_det0_trigger_level	1			#the level signal of  outage detection E0 module (1:high level;0:low level)
>cfg_filament_det1_trigger_level	1			#the level signal of  outage detection E1 module (1:high level;0:low level)

>cfg_filament_load_length		100			#the lenght to extrude filament (mm),Max:2000mm
>cfg_filament_load_speed		800			#the speed to extrude filament(mm/min)
>cfg_filament_load_limit_temperature	200			#It is the minimum temperature to extrude filament .

>cfg_filament_unload_length		100			#It is the minimum temperature to extrude filament .
>cfg_filament_unload_speed		800			#the speed to retract filament(mm/min)
>cfg_filament_unload_limit_temperature	200			#It is the minimum temperature to retract filament .

#=========================== Bed Leveling  =============================
#Leveling mode settings
>cfg_leveling_mode			1			#1:auto-leveling; 0:manual leveling

#=========================== Manual Leveling ==================================

>cfg_point_number			5			#the point number of manual leveling(3,4,5 point available)

#the 5 point location of  manual leveling
>cfg_point1:50,50						
>cfg_point2:280,50
>cfg_point3:280,280						
>cfg_point4:50,280
>cfg_point5:130,130

#========================== Auto Leveling ===================================

>cfg_auto_leveling_cmd:G28;G29;                                #the setting of auto-leveling command button

#=================== Language settings ==========================================
>cfg_language_adjust_type		1	#multi-language(enable:1, disable:0)

>cfg_language_type			2	#languages setting,this configuration is valid when "cfg_multiple_language" is disabled. 
						#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).
#==================Display Effect(refer to MKS TFT datasheet)=================================
#user-defined function1
>setmenu_func1_display		1				#1:display this buttonŁ¬0:no

#each command must be separated by semicolon ";"
>setmenu_func1:M84;

>moreitem_pic_cnt		0				#the number of "more" button 

#edit command for 1~7 "More" button
#each command must be separated by semicolon ";"
#each icon command must be less than 200 bytes
>moreitem_button1_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button2_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button3_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button4_cmd:G28 X0;G28 Y0;G28 Z0;

[... 234 lines cut ...]
>CFG_WIFI_CLOUD_HOST		www.baizhongyun.cn	#cloud server url
>CFG_CLOUD_PORT			10086			#cloud server port
>WISI_LIST_SCAN                 1                       #Whether to use wifi list scan (1: use, 0: forbidden)

#Do not display wifi button 0: Display WiFi button 1: Do not display
>DISABLE_WIFI		1

#About information
>about_type:Robin_nano35
>about_version:V1.0.3
>about_company:SP_electronic
>about_email:pepe1125@gmail.com

#Suspend/resume extrusion configuration (unit: mm)
>PAUSE_UNLOAD_LEN   -3
>RESUME_LOAD_LEN    10

#Suspend/resume extrusion increase percentage, when 0, the item does not work (unit: %)
>RESUME_SPEED    80

#Wake mode configuration
>cfg_Standby_mode		0			#Whether to enable the wake-up function (0: off, 1: on)
>cfg_Standby_gap_time           600                     #Set to enter sleep time (unit: second)

#end.

makerbase-mks, 2020-03-05:
[image]
These two values you set to the same value try.

mks-viva, 2020-11-11:
Hi! No response, then I will temporarily close this question.

closed 2020-11-11 by mks-viva