makerbase-mks/MKS-Robin-Nano-V1.X #6: TFT35 screen on Sapphire PRO

EG1313, 2020-01-27:
Have installed Marlin 2 ("bugfix") on Sapphire Pro and it works fine (board MKS Robin Nano).
Trying to do the same with latest Marlin release (official), ATOM produces the BIN file but as on upload screen goes to 320x240 resolution and so far didn't manage to bring it to 480x320 pixel size.
Upscale x3 - doesn't work, low resolution menu screen with clearly visible items is jammed to upper left corner.

Please help, thanks.
[image]

makerbase-mks, 2020-02-29:
I has upload Nano+robin tft35 marlin firmware
[image]

EG1313, 2020-03-29:
Hello!
my point is - I have succeeded with Marlin 2.0 (bugfix) version but no luck with official releases - any of them. 
For instance I'm trying to get screen "in place" with 2.0.5 and having the same issue.
Permanently low resolution screen gets attached to upper left corner, so called "upscale 3" wont work. 
One more, buttons are working and they are where they are suppose to be but hidden.
Essentially printer is functional but screen...

Appreciate any help on this, thanks!
Attaching my configuration files  

P.S. as well, I can control the printer with any terminal program.

[image]
[Marlin_conf.zip](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4398283/Marlin_conf.zip)

mks-viva, 2020-11-11:
Hi! At present, the latest marlin firmware has fully supported this function. So I will close this question.

closed 2020-11-11 by mks-viva