makerbase-mks/MKS-Robin-Nano-V1.X #7: Random movements during printing due to sd card reading errors

atomphil, 2020-02-02:
I got an issue with the MKS Robin nano v1.1. The print head moves to random locations in the middle of printing, after that printing continues as normal.
I have activated the debugging of the board (M111 S7) and can see that the errors happen randomly when reading from SD card. It is not because of the SD card itself. I have tried different tested cards and different file systems.
At the moment I use vanilla Marlin.

In the change log there is a point:

    2.V1.0.2 Modifications:
    2.1.Fix the bug that read Large capacity sd card data error sometimes.

@makerbase-mks What did you do to fix the read errors?

makerbase-mks, 2020-03-05:
I modified some card reading functions and changed the types of some of these variables

atomphil, 2020-03-05:
> 
> 
> I modified some card reading functions and changed the types of some of these variables

What did you changed to make it work better?

makerbase-mks, 2020-03-05:
[image]
You can change the type of the variable so that it can accommodate a larger address.

mks-viva, 2020-11-11:
Hi! I will close this question temporarily.

closed 2020-11-11 by mks-viva