makerbase-mks/MKS-Robin-Nano-V1.X #8: Adding a language

karohan, 2020-02-24:
Hello.

I want to add the language of my country.
I cannot edit the .bin files.
Markerbase Image Editor Does not work properly.

[image]

Can you help me?

makerbase-mks, 2020-02-25:
Hello, this tool can help you.
https://github.com/makerbase-mks/Software/blob/master/MKS_Image2bin_tool/MKS_Image2bin_tool.rar

karohan, 2020-02-25:
I know how to make .bmp to.bin files. 
I did not ask you this.
how to add a new language?
Please help me

makerbase-mks, 2020-02-25:
Hello, there are two ways to add a language:
1. Disable multi-language in the configuration file
> cfg_language_adjust_type 0 # multi-language (enable: 1, disable: 0)
You add your language when designing the picture, and then convert the picture into a .bin file. After updating, you can see the text you designed on the picture.
2. The source code needs to be modified. We have opened the source code for Robin_nano35. You can modify the source code to add your language.

karohan, 2020-02-26:
hi mks.
I managed to turn the buttons.
where should i translate the articles?
I can't interfere in Chinese writing
All posts, error messages, alerts are in chinese
Please Help me. 

[image]

[image]
[image]
[image]

makerbase-mks, 2020-02-27:
The articles supports the following 7 languages, which you can configure in the configuration file. If you want to add other languages, you need to modify the source code:
> cfg_language_type 2 #languages setting, this configuration is valid when "cfg_multiple_language" is disabled.
# (simplified Chinese: 1; traditional Chinese: 2; English: 3; Russian: 4; Spanish: 5; French: 6; Italian: 7).

karohan, 2020-02-27:
Thank you. Where can I find the source code?

makerbase-mks, 2020-02-27:
https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware/tree/1.0.5

makerbase-mks, 2020-02-27:
https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware/tree/1.0.4

karohan, 2020-02-27:
[image]
I did not make any changes.
I wanted to test it.
I get this error.
thanks a lot for your support.
All my devices mks <3

makerbase-mks, 2020-03-05:
[image]
Maybe you should open it with this version.

karohan, 2020-03-12:
[image]

What is the problem?

mks-viva, 2020-11-11:
Has this problem been resolved?

closed 2021-08-09 by mks-viva