makerbase-mks/MKS-Robin-Nano-V1.X #10: No manual move if endstops on double Z triggered

PowerEditon, 2020-03-03:
Hi,

with double Z only Z,
double optical/mechanical endstops,
endstops on Z **triggered** after homing,

power off,
restart,

now you can't move any axis over manual menu,

homing on all axis are working!

Simple solution is to move Z 5 mm after homing to untrigger, but i cloud not 
found a setting for the config file.

Thanks

PowerEditon, 2020-03-03:
`#===================Advanced  Settings=======================================
>cfg_insert_det_module			0			#Assemble power outage module (1:mks 220det;0:mks pwc)

>cfg_have_ups_device			0			# UPS(1:yes;0:no)

>cfg_print_over_auto_close		1			# auto-off when print finishes(1:yes£ª0:no)

>cfg_filament_det0_trigger_level	0			#the level signal of  outage detection E0 module (1:high level;0:low level)
>cfg_filament_det1_trigger_level	0			#the level signal of  outage detection E1 module (1:high level;0:low level)

>cfg_filament_load_length		100			#the lenght to extrude filament (mm),Max:2000mm
>cfg_filament_load_speed		800			#the speed to extrude filament(mm/min)
>cfg_filament_load_limit_temperature	200			#It is the minimum temperature to extrude filament .

>cfg_filament_unload_length		100			#It is the minimum temperature to extrude filament .
>cfg_filament_unload_speed		800			#the speed to retract filament(mm/min)
>cfg_filament_unload_limit_temperature	200			#It is the minimum temperature to retract filament .

#=========================== Bed Leveling  =============================
#Leveling mode settings
>cfg_leveling_mode			0			#1:auto-leveling; 0:manual leveling

#=========================== Manual Leveling ==================================

>cfg_point_number			5			#the point number of manual leveling(3,4,5 point available)

#the 5 point location of  manual leveling
#Printer Head Body at the corners of the headbed, X,Y
>cfg_point1:30,295   #BL (Firmware 1.02 - 1.03)
>cfg_point2:280,295  #BR
>cfg_point3:280,25   #FR
>cfg_point4:30,25    #FL
>cfg_point5:155,150  #Center

#========================== Auto Leveling ===================================

>cfg_auto_leveling_cmd:G28;G29;                                #the setting of auto-leveling command button

#=================== Language settings ==========================================
>cfg_language_adjust_type		1	#multi-language(enable:1, disable:0)

>cfg_language_type			3	#languages setting,this configuration is valid when "cfg_multiple_language" is disabled. 
						#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).
#==================Display Effect(refer to MKS TFT datasheet)=================================
#user-defined function1
>setmenu_func1_display		1				#1:display this button£¨0:no

#each command must be separated by semicolon ";"
>setmenu_func1:M84;

>moreitem_pic_cnt		0				#the number of "more" button 

#edit command for 1~7 "More" button
#each command must be separated by semicolon ";"
#each icon command must be less than 200 bytes
>moreitem_button1_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button2_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button3_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button4_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button5_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button6_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button7_cmd:G28 X0;G28 Y0;G28 Z0;

#the number of "more" button to dispaly on printing interface
[... 207 lines cut ...]
>DELTA_SMOOTH_ROD_OFFSET	211.5	#Horizontal offset from middle of printer to smooth rod center.
>DELTA_EFFECTOR_OFFSET		28	#Horizontal offset of the universal joints on the end effector.
>DELTA_CARRIAGE_OFFSET		14.5	#Horizontal offset of the universal joints on the carriages.
>DELTA_RADIUS			169	#Horizontal distance bridged by diagonal push rods when effector is centered.	
>DELTA_HEIGHT			302	#height from z=0.00 to home position
>DELTA_PRINTABLE_RADIUS		125	#Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
>DELTA_CALIBRATION_RADIUS	100	#set the radius for the calibration probe points - max 0.8 * DELTA_PRINTABLE_RADIUS 

#============================== Wifi Settings =============================
>CFG_WIFI_MODE			0			#wifi mode(0:sta;1:ap)
>CFG_WIFI_AP_NAME		WiFitest		#wifi name
>CFG_WIFI_KEY_CODE		makerbase		#wifi password

>CFG_CLOUD_ENABLE		1			#cloud service enable(0:disable;  1:enable)
>CFG_WIFI_CLOUD_HOST		www.baizhongyun.cn	#cloud server url
>CFG_CLOUD_PORT			10086			#cloud server port

#============================== About information =========================

>about_type:Robin_nano V1.2
>about_version:V1.0.4
>about_company:PE
>about_email:

#end.`

bates1337, 2020-03-20:
Same probleme here with a mks robin nano v1.1

MKS-Sean, 2020-03-28:
[robin_nano35_cfg.txt](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4396565/robin_nano35_cfg.txt)
I use this configuration file is normal, you can modify some parameters, such as pulse, motor direction, etc., and then try to use on your machine