makerbase-mks/MKS-Robin-Nano-V1.X #11: Firmware v2.x.x for nano + tft32

artdezigner, 2020-03-15:
Hello, please tell me, the firmware v.2.x.x will be for Robin nano v1.1 + TFT 32? 
Can I use v2.0.0 firmware for Nano+tft 35 for my configuration?

And second question: does the pin of coldend fan work automatically in firmware v1.0.4?

mks-viva, 2020-03-16:
Hi! 
1. firmware v2.x.x default is Nano+Robin TFT35
2. This FAN is in the printing process, it will be automatically turned on after the first layer printing is completed

artdezigner, 2020-03-17:
please tell me when the firmware for Nano+TFT 32 will be made?

duyleekun, 2020-05-23:
I'm using the tft24, where to change the config?

mgcrea, 2020-06-13:
@mks-viva I've seen that there is a `Robin-TFT32` git branch on the [firmware repository](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/Robin-TFT32), would this branch work with a TFT32 or is it still work in progress?