makerbase-mks/MKS-Robin-Nano-V1.X #12: Mks_Nano35_V2.0.1 can't use G34 - Z Steppers Auto-Alignment

chenxiaowei107, 2020-04-09:
the next version firmware will Support？
My machine have dual Z and bltouch, If i use marlin2.0,I can user Z steppers Auto-Alignment ,It's very convenient and convenient。But marlin is ugly and not convenient for 3.5TFT Touch-screen。
The MKS-Robin-Nano was Very convenient and beautiful，I like it.
I Hope Next Firmware will support Z Steppers Auto-Alignment.

makerbase-mks, 2020-04-10:
Hi! V2.0.1 version firmware, already compatible with this function.

chenxiaowei107, 2020-04-10:
> Hi! V2.0.1 version firmware, already compatible with this function.

How can I use?

chenxiaowei107, 2020-04-10:
> Hi! V2.0.1 version firmware, already compatible with this function.
I send G34 code ，but the machine no response

makerbase-mks, 2020-04-10:
Of course! You can download it and update it to your Nano board.
[image]
https://raw.githubusercontent.com/makerbase-mks/MKS-Robin-Nano/master/Nano%20firmware%20v2.x.x/Mks_Nano35_V2.0.1%20Release%20file.rar

chenxiaowei107, 2020-04-10:
> Of course! You can download it and update it to your Nano board.
> [image]
> https://raw.githubusercontent.com/makerbase-mks/MKS-Robin-Nano/master/Nano%20firmware%20v2.x.x/Mks_Nano35_V2.0.1%20Release%20file.rar

I have Flash this firmware，but the menu no  Z steppers Auto-Alignment menu.

makerbase-mks, 2020-04-10:
This function of G34 may not be supported, but 3DTOUCH and limit function are compatible.

chenxiaowei107, 2020-04-10:
> This function of G34 may not be supported, but 3DTOUCH and limit function are compatible.

我是不是回复中文就可以了，marlin2.0可以利用bltouch自动平衡z轴。2.0.0.1的固件没有这个功能。

makerbase-mks, 2020-04-10:
2.0.1固件已经支持bltouch和限位功能，但是还不支持G34指令

chenxiaowei107, 2020-04-10:
> 2.0.1固件已经支持bltouch和限位功能，但是还不支持G34指令

是的，支持bltouch我知道，但是希望下个版本可以支持G34，我的是双z，用e1电机位置作为第二个z轴，marlin2.0 G34可以自动校准2个z轴的高度，非常方便实用。

makerbase-mks, 2020-04-10:
这个可以有，后续我们会考虑增加上去。

chenxiaowei107, 2020-04-10:
> 这个可以有，后续我们会考虑增加上去。

因为用双z，所以现在只能用marlin，但是marlin真的界面太难看，也不友好，mks的固件比较方便。期待你们得下个版本。

hakimio, 2020-08-11:
@makerbase-mks any news on G34 (Z axis Auto-Alignment)?