makerbase-mks/MKS-Robin-Nano-V1.X #14: Compiling for Robin Nano35 issue

imberest, 2020-04-21:
Hello, I've been trying to tweak Marlin 2.0 for my needs, but it fails to compile. I've tried stock Marlin 2.0 and it compiles with no problems.
Mks-Robin-Nano-Marlin2.0-Firmware-master gets stuck on:
**Compiling .pio\build\mks_robin_nano35\lib22d\USBComposite\usb_scsi_data.c.o
Compiling .pio\build\mks_robin_nano35\lib22d\USBComposite\usb_setup.cpp.o
Compiling .pio\build\mks_robin_nano35\lib22d\USBComposite\usb_x360.c.o
Archiving .pio\build\mks_robin_nano35\lib22d\libUSBComposite.a
Archiving .pio\build\mks_robin_nano35\lib690\libU8glib-HAL_ID1932.a** <--- stuck...

Tried VS+PlatformIO and tried MKS-Marlin2.0-Compile-Tool so far.

imberest, 2020-04-22:
So, the new release fixed it and I can successfully compile it. The issue I am having is that TFT35 is blank.
Does anyone have any tips or ideas?

mks-viva, 2020-04-26:
Hi! You can use mks has add UI marlin2.0.5, link：https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

mks-viva, 2020-11-11:
Hi! No response, then I will temporarily close this question.

closed 2020-11-11 by mks-viva