makerbase-mks/MKS-Robin-Nano-V1.X #15: TMC2208 UART Connection Error

Leetspeak1286, 2020-04-24:
382/5000
Hi,

I try to run longer on the Robin Nano 1.2 TMC 2208 in UART mode but always get

X 0x00: 00: 00: 00 Bad response!
Testing X connection ... Error: All LOW

Have I tried 1 cable like a Y cable with 1k ram stand for TX and always the same?

Uploaded my Pins.h and Configs, unfortunately I can't get any further: /
[conf.zip](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4531436/conf.zip)

closed 2020-04-26 by le3tspeak

VIXIVIXIV, 2020-05-05:
@le3tspeak, were you able to get it to work? I'm planning on a driver upgrade and want to UART.

hannesweisbach, 2020-12-11:
Have you reduced the baudrate? 19200 worked for me.

I put `#define TMC_BAUD_RATE 19200` in `Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h`. I think the default is something like 56700 and that was too fast to reliably work. It worked sometimes but the my TMC2208 did not always initialize properly and M122 never worked. However I could see the TMC2208 sometimes responded to a status request from Marlin on the oscilloscope.

thierryzoller, 2020-12-11:
This was closed, but what was the solution?

hannesweisbach, 2020-12-11:
@le3tspeak never commented on what was the problem or if s/he solved it. From my experience I'd say too high of a baud rate in the config. But this is just a guess on my part at this point.