makerbase-mks/MKS-Robin-Nano-V1.X #16: BL-Touch 3.1V does not work properly

heavy2119, 2020-04-25:
Wiring is normal

Problem.

1. Power on → BL touch pin needle up and down 2-3 times → pin needle waiting
→ When the signal line is connected, the power turns on → The pin needle comes out
→ Remove signal wire and turn on the power → Pin needle does not come out

2. It does not work on M280 P0 S90, M280 P0 S10, but it works up / down using M280 P0 S120.

3. Pin-up, pin-down and pin-up are not detected during auto-leveling.
→ Auto leveling results are well maintained.

Attach the video in question

https://youtu.be/2M8nTG85QXc

https://youtu.be/myXfHmgcO8k

mks-viva, 2020-04-26:
Hi! V2.0.0 firmware version has support bltouch v3.1, you can download it and test...

closed 2020-05-21 by makerbase-mks

Bunta-Impreza, 2020-09-27:
Hello, 
News? I have the same problem with BLT 1.3 and 3Dtouch.
Version 3.0.3.3