makerbase-mks/MKS-Robin-Nano-V1.X #17: Bltouch does not work on 2.0.0v and 2.0.1v

heavy2119, 2020-04-26:
Bltouch does not work on 2.0.0v and 2.0.1v
I want to find the cause
Please help

AuntJemima, 2020-04-28:
> Bltouch works on 2.0.2v but Servo does not rise after probe detection

I have this exact problem as well, and it's the only thing preventing me from printing with the BLTouch!

james013, 2020-05-09:
works for me:

in config file :

>Z_MAX_ENDSTOP_INVERTING 1

>Z_MIN_PROBE_ENDSTOP_INVERTING 1

>USE_ZMAX_PLUG	1

>Z_MIN_PROBE_PIN_MODE		2	# 0 : NULL; 1: ZMIN; 2: ZMAX

>BLTOUCH			1	# 0:Disable BLTOUCH; 1:Enable BLTOUCH

AuntJemima, 2020-05-09:
> 
> 
> > USE_ZMAX_PLUG 1

Perhaps the problem is that I put it in ZMIN rather than ZMAX? I'll have to give that a shot instead.

Bunta-Impreza, 2020-07-08:
Hello, any news here?
I have the problem too.
I’m on version 2.0.3.2
[robin_nano35 Bunta_cfg.txt](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4890866/robin_nano35.Bunta_cfg.txt)

https://www.youtube.com/watch?v=MFSIoimxCeg&feature=youtu.be

Sorry for my poor English, I use reverso.