makerbase-mks/MKS-Robin-Nano-V1.X #18: MKS Robin Nano TMC 2209 v 3.0 how is install ?

dejavuxer, 2020-05-13:
Hi friend. 

Sorry for my broken English. I have two trees bluer 3d printer.
There are 2 TMC 2208 and 2 A4998 on it. Can I replace it with the TMC 2209 v3 uart I have? 

Can you tell me how to do it? 
Can you give information about the software and installation. 
Should I take action on the config file?

Thank you.

Teslafly, 2020-05-20:
The uart pins are not connected correctly to the pins on the tmc2209 drivers. I am also trying to do this. 
you will need to use the wifi (or other) uart port and run it to the tmc drivers configured on different addresses using the microstepping jumpers, the uart will be connected to the slp pins direct to rx, and to tx through a 1k resistor.
I may post here if i get my setup working. 

this would not be hard to implement in a hardware revision. 
If they did, i would suggest adding jumpers to support native usb too.

makerbase-mks, 2020-05-21:
1. You can use an IO to drive the TMC2208 / 2209 UART mode; UART_TX = UART_RX
2. You can connect to the free pin of the WIFI module to drive the TMC UART

GrumpaV2, 2020-05-22:
Good so WiFi will still work then. Has this been tested?

makerbase-mks, 2020-05-26:
Because you use marlin2.0, robin wifi is not available, so wifi IO can be used for TMC uart

closed 2020-06-06 by makerbase-mks