makerbase-mks/MKS-Robin-Nano-V1.X #21: cfg_Standby_mode and hesitation problem

serdartaylan, 2020-05-16:
Hello there.
I use "Two Tress" brand "Bluer" model printer.
My printer now has v2.0.2 installed. I use "Cura".

I set the cfg_Standby_mode value to 1 to turn off the screen from the settings. My screen turns off at the end of the time I set.

Sometimes, when I touch the screen to turn the screen back on, there is a pause of 1 second (maybe shorter) during the printing process.

My Experience:
This pause does not make the process of filling.
It usually pauses when building the outer walls.

What is the problem, how should I solve it?

Do you want me to shoot a video?

makerbase-mks, 2020-05-21:
Hi! If you do n’t set it, will the printer not be suspended?

serdartaylan, 2020-05-21:
If I turn off the "cfg_Standby_mode" setting, that is, if the screen is always on, there is no hesitation.
this happens only when we touch the screen after the screen turns off at the end of the time.
I can shoot videos if you want...

cemiloztas, 2020-05-21:
What do you think Mks ? You are asking to user. Did not you test your card?  when it was wake up from stand by mode what was happening and how is it wake up . Is there any problem etc.

ttutuncu, 2020-05-27:
There is a problem in the 2.0.2 version, maybe older versions too. When the screen is turned off by standby mode we need to wake up the screen by touching it. If the printer is printing a wall and we touch the screen to wake it up the printer pauses. This pause causes a blob on the wall. Can you check the standby modes behaviour while the printer is printing. There is some sort of lag causing the board to stop sending gcode to the printer. This mostly happens on wall prints, it doesn't occur while printing infill.