makerbase-mks/MKS-Robin-Nano-V1.X #22: Excessive heating of the motors. (In this case only the Y axis)

LuigyU, 2020-05-26:
After waiting a long time due to the delay with the shipping of the covid-19, I finally received the item.
Then I installed it on my printer, following all the instructions I received from its github portal and with great satisfaction to see that everything was working correctly.

But my disappointment was that during the first impression I noticed that the nema17 0.9A motor of the Y axis was overheated (to such an extent that it could not be touched by hand because it burned, it was almost like the bed at more than 70º or more )
The X and E0 axis motors were also warming up but were acceptable.

Firstly, see if my Y axis car had any impediment in its movement that made it go wrong and it was not.

I understood that the voltage with which the various MKS TCM2208 V2 came as they come from the factory is too high (1,171) and I proceeded to lower it as I had previously on my old board. (0.80V)

Still, it was still very hot, and I proceeded to lower it even further, to the 0.7 limit of the driver.
The temperature of the Y motor was hot and seemed correct but in this situation, I was already missing my motor steps.

Believing that the driver could be damaged, I changed it with another, but it was still just as hot.

Perhaps suspecting that my motor might be damaged, I would change it for a different and stronger one, a 1.2A nema 17 and adjust the driver to 1V to test it. I had to stop printing (which was an hour) because the engine was excessively hot.

All my mounting information that I have followed step by step and the software I have found here.
https://github.com/makerbase-mks/MKS-Robin-Nano/tree/master/Nano%20firmware%20v2.x.x
and from the videos
https://www.youtube.com/watch?v=_qMahRSUtDY
Is what I have done correct?

The X, Z, and E0 axes are fine, but I **only have the problem on the Y axis.**

I don't know what to do anymore, I have everything stopped and I can't print anything without compromising the integrity of the printer.
I really need your help and advice

chshkhr, 2020-06-11:
Hi,
correct `Vref `for [TMC2208](https://wiki.fysetc.com/TMC2208/) is **1,69V**.
While correct `Vref` for [TMC2209](https://wiki.fysetc.com/TMC2208/) is **2,82V**.
`Vref` is proportional to `Irms` (as for TMC2008 `Vref = (Irms * 2.5V) / 1.77A = Irms * 1.41 = Imax`). So the more `Vref` you set the less current you need to get the same power `P = I * V`. So to reduce consumed current (which is the reason for motors heating) you need to increase`Vref`. 
Am I right?
Anyway, I increased `Vref` for my TMC2209 up to 2,61V (this is the maximum value I could get via the resistor on TMC2209). Temperature decreased from 84C to 50C while printing. Heating on standby mode (without disabling motors) disappeared at all

thierryzoller, 2020-06-11:
Sounds counterintuitive to me, I am interested in the responses.

chshkhr, 2020-06-11:
As I understand now we should look at not only on RMC of stepper driver but on RMC of stepper motor because it much lower... _"For NEMA 17 motors from 3D printers the current is in general in the range of 0.5A to 0.8A RMS, which is a reference voltage (Vref) of 0.7V to 1.1V."_
[Here is the source](https://learn.watterott.com/silentstepstick/faq/)