makerbase-mks/MKS-Robin-Nano-V1.X #26: Editing the Russian language

Engi23, 2020-05-30:
1. Replace the "выдавить" to "Извлечь". The word "выдавить" in Russian is associated with squeezing plastic from a nozzle, and introduces confusion and confusion.
It is also advisable to replace "втянуть" на "Заправить"
2. Correct errors in words
"Ткорость экстр" right  "Скорость экстр"
"Ткорость движ" right  "Скорость движ"
3. Can you write words with a capital letter? It's a shame, in all languages words are written with a capital letter, and in Russian with a small one. If you need, I will provide a complete list of words written from a large buva, if you have any problems with typing.

makerbase-mks, 2020-06-10:
Thank you for your feedback. Can you list the problematic translations or use improper vocabulary, we will correct it.

makerbase-mks, 2020-06-16:
Ok thank you again!

Spider84, 2020-06-18:
May be make font change to more better?