makerbase-mks/MKS-Robin-Nano-V1.X #29: Inductive sensor options?

themacboy, 2020-06-11:
It is possible to use a inductive sensor on a Robin Nano board?

How can I do that?

makerbase-mks, 2020-06-16:
As long as your module meets the output high and low levels

Engi23, 2020-06-21:
I used this circuit
[https://www.cnc-club.ru/forum/download/file.php?id=24409&mode=view](url)

themacboy, 2020-06-28:
I dont need all that, Im using a 5v inductive sensor from trianglelabs:
https://es.aliexpress.com/item/33006152423.html?spm=a2g0o.productlist.0.0.4d7443fchmOY7J&algo_pvid=fe75920c-5e00-4ffb-a765-c10012d45c16&algo_expid=fe75920c-5e00-4ffb-a765-c10012d45c16-0&btsid=0b0a119a15933768323478328eaf45&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

The current problem is that after instgallation Im not sure what parameters I must set in config, and the motherboard turn crazy and set Z displaced 40 mm and Z servo stop working when I set autolevel to on.

Can someone help with configs? Thanks in advance

themacboy, 2020-06-28:
here example of my configs and the problem:

https://www.youtube.com/watch?v=P_9MJ-f3pHE

Engi23, 2020-07-15:
1. The offset does not go there, the printer does not enter the table with the sensor. Try an offset of X -100 instead of 100.
2. Naturally, you need to check that the sensor sees metal - the lamp on the sensor lights up.
3. It is advisable to check that the printer sees the signal from the sensor, for example by connecting it instead of Z home. Perhaps the sensor is incorrectly electrically connected or the type in the firmware is incorrectly selected (open / close)

mks-viva, 2020-11-11:
@themacboy 
Has this problem been resolved?

themacboy, 2020-11-12:
yes, thx I have learned a lot since this question.

closed 2020-11-12 by themacboy