makerbase-mks/MKS-Robin-Nano-V1.X #30: TFT35 screen

usebaz, 2020-06-11:
Hi,
I want to try add support of TFT35(Robin TFT 3.5, Sapphire Pro) with Robin Nano to Klipper. When I can find more info about screen(type of microscheme, maybe some docs how work with it)?
Thanks

makerbase-mks, 2020-06-16:
Hi! Do you want to know information about Robin TFT35, such as the driver IC of the screen, connection method, initialization...

usebaz, 2020-06-16:
I need realize similar function: https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/display/st7920.py 
and for it I want as much information as possible about Robint TFT35. Thanks!

mks-viva, 2020-10-22:
MKS Robin TFT35:
1. FSMC 16bit 
2. use ILI9488
3. 320*480
About inital code, you can refer to : https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

mks-viva, 2020-11-03:
I will close this question temporarily

closed 2020-11-03 by mks-viva