makerbase-mks/MKS-Robin-Nano-V1.X #32: MKS Robin Nano boot loader

haxiny, 2020-06-17:
Hi, i need MKS Robin Nano BOOT LOADER. Does anyone have it?

Spider84, 2020-06-18:
I have from nano 1.1 with TFT3.2

haxiny, 2020-06-18:
> I have from nano 1.1 with TFT3.2
Please....Can you upload it? Thank you very much! I need it!

Spider84, 2020-06-18:
[MKS-Robin-Nano-Bootloader.zip](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4798525/MKS-Robin-Nano-Bootloader.zip)
This loader don't flash nano35 firmwares...
Let me know if you can find TFT35 board owner :) I'll show way how to download bootloader from work board.

haxiny, 2020-06-18:
OK!Thank you very much!
I will find it!

nickrodie, 2020-06-30:
> [MKS-Robin-Nano-Bootloader.zip](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4798525/MKS-Robin-Nano-Bootloader.zip)
> This loader don't flash nano35 firmwares...
> Let me know if you can find TFT35 board owner :) I'll show way how to download bootloader from work board.

I'd be happy to help. It won't be a problem I've downloaded boot loader before. Just can't remember exactly how so instructions required. Nano V1.2 TFT 35

Spider84, 2020-06-30:
Download and compile this firmware https://github.com/Spider84/MKS-Robin-Bootloader (or I can give compiled)
Flash it like you flasing regular firmware. And Wait some time (10-15 sec), after take SDcard - it will contain bootloader.bin file.
After this you can flash back default firmware.

nickrodie, 2020-07-01:
Flashed bootloader. Got stuck at end of TFT load 100%. No file saved to SD.

Spider84, 2020-07-01:
Check that SD connected to same pins and spi1.
Alsow try connect to USB-uart and chack "log" in console.

nickrodie, 2020-07-01:
Nah. It doesn't work. The documentation is all available.

Spider84, 2020-07-01:
you flashed only bootloader? Which one?

nickrodie, 2020-07-01:
I built the nano35 from your source

Spider84, 2020-07-01:
Can you connect USB cable to you printer and open terminal on pc immediately after turning printer on with 115200 bps?
And wait 10-15 seconds. It must print STM32F103ZET6 and then error if error ocured.

nickrodie, 2020-07-01:
I would hope it didn't I have a VET chip

Spider84, 2020-07-01:
[Robin_nano.zip](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4856995/Robin_nano.zip)
Can you test this?

nickrodie, 2020-07-01:
Ok that one worked.
Was a bit too tired yesterday. Had a fresh look this morning.
If I compile the source I notice my files are about 28kb. Not sure why they larger than yours. I get nothing back from console.
I ran your original compiled version with console and it failed because couldn't get sd access. I noticed it was built for ZET6 though.
Tried the latest version you sent. Success :) 
[Robin_nano35_bootloader.zip](https://github.com/makerbase-mks/MKS-Robin-Nano/files/4861259/Robin_nano35_bootloader.zip)

nickrodie, 2020-07-01:
FYI fan didn't rotate. I have bare board with nothing but screen attached atm.

mks-viva, 2020-11-03:
[image]
I will close this question temporarily.

X-Ryl669, 2021-01-07:
I've set up a [blog post](https://blog.cyril.by/en/3d-printers/mks-robin-nano35-advanced-debugging) about how to use this bootloader after unlocking it so you can debug the board a lot more easily

X-Ryl669, 2021-01-07:
@mks-viva The bootloader's HEX file in this repository does not work: it's 1.2MB, you can't program it on the 512kB flash, or I'm missing how you do that.

mks-viva, 2021-01-07:
OK, I need to report this issue to the engineer.

poon-holder, 2022-03-21:
Why are the robin boards so quirky?  X-Ryl blog is very informative. many thanks for putting that together. 
I have had a huge headache flashing a v1.1 and now v1.3 both got stuck at "Updating 100%" often changing the .bin file to the correct extension. however the do not properly flash and will not communicate over usb with host. It seems only trial and error works with these boards...?