makerbase-mks/MKS-Robin-Nano-V1.X #36: Flashing firmware via MKS Slot2 fails

Miraculix200, 2020-06-19:
I have a problem with my MKS Slot2 SD card extension connected to MKS Robin Nano v1.2.

When I try to flash a firmware with the SD card connected to MKS Slot2, it says "TFT flashing" on the display and jumps from 1% straight to 100%. But the firmware is still the same as before.

When I flash it through the onboard SD card slot, then everything works fine. 

Is this normal? Or is there anything I can do about it?

Note that I cut the MKS Slot2 cable to make it about 20cm longer. But my MicroSD to SD card extension has about the equal length and doesn't create that problem.

makerbase-mks, 2020-07-08:
The line is same as?

Miraculix200, 2020-07-08:
Yes, it's the same. Just longer.

Are the lines the same on the board? MKS Slot2 connector and SD card connector have the same lines?

mks-viva, 2020-12-01:
Hi! This phenomenon is that the cable is too long, which leads to an error in reading the SD card firmware. It is recommended to use a short cable.

closed 2020-12-01 by mks-viva