makerbase-mks/MKS-Robin-Nano-V1.X #38: White screen on when i printing

pablorusso88, 2020-06-24:
hi!

i have a two tree printer on fw 2.0.1

why when I'm printing my screen turns white?
I touch the screen and it doesn't come back, but it does respond because sometimes I touch something and it stops printing

thanks!
i new on this

n0xxA1337, 2020-08-02:
I have the same problem on Mks_Nano35 V2.0.3.3. Are you printing from Octoprint or the SD card ?

pablorusso88, 2020-08-03:
SD card, the fw 2.3.1 works fine

makerbase-mks, 2020-08-11:
You can try to update firmware to latest version

wealthybigpenis, 2021-01-23:
i am also having this problem with 2.03.5. it only happens about about an hour into printing once the display has gone to sleep. when i touch it to wake it up i have a white screen the rest of the print

mks-viva, 2021-01-25:
@wealthybigpenis 
Can you provide some picture to me when you printing?

wealthybigpenis, 2021-01-27:
[image]
Just a solid white screen. Been printing about 3 hours.

mks-viva, 2021-01-29:
Is printing still in progress? Just the screen cannot display information?

wealthybigpenis, 2021-01-29:
Yes the printing continues and finishes however the printer has to be power cycled before the screen will start working again

mks-viva, 2021-02-01:
What seems to affect the display

rubiotel, 2021-02-09:
Exactly the same thing happens to me, after 2 or 3 hours of printing, I have to turn off the printer and when I try to load filament (outwards) the Z rises.

iMarkus, 2022-11-14:
Noticed the same on a TwoTrees Bluer printer with Marlin 2.1.1
Sometimes this happens directly after starting the printing process via OctoPrint.

Printing process is not affected.