makerbase-mks/MKS-Robin-Nano-V1.X #40: File upload via wifi error

nickrodie, 2020-06-30:
When I try to upload gcode via wifi from pronterface it fails with
`[ERROR] Can't write to printer (disconnected?) (Socket error 10054): An existing connection was forcibly closed by the remote host`
Can anyone help?

Nano V1.2 
TFT 35 
Mks_Nano35_V2.0.3.2

Engi23, 2020-07-06:
I use ultimaker cura, Mks_Nano35_V2.0.3.1 it work. Try to place the printer not far from wifi. microSD inserted? The problem is not fully described. Is the printer connected to the wifi? Does the program see the temperature of the extruder, etc.?

mks-viva, 2020-12-01:
I think your plug-in should be updated to the latest. You can check the wifi connection status on Robin TFT35, etc.

mks-viva, 2021-01-18:
Hi! If have not response, i will close this issues temporarily.

closed 2021-01-18 by mks-viva