makerbase-mks/MKS-Robin-Nano-V1.X #41: linear Advance

jan-mike, 2020-07-04:
Add Linear Advance on LCD setting on latest firmware

Please maybe you can ADD linear Advance on latest firmware so that it is easy to adjust on LCD Display..

Thank you

Engi23, 2020-08-11:
I ran a test with a k factor from 0 to 2 and I don't see any difference. Apparently the Linear Advance is off. 
You can at least turn on Linear Advance? Please...

jan-mike, 2020-08-11:
I have no idea how to enable the linear advice. Maybe some can add the
linear advance seeting on the display setting so that no need to configure
on firmware.

Jan

RFSTER, 2021-06-22:
In order to use Linear Advance, your steppers need to be operating in UART mode. By default they are using Step/Dir mode. This requires hardware mods if you are using a MKS Robin Nano V1.1 or V1.2. On other more recent boards, it is a matter of setting the jumpers under the stepper driver boards correctly. You also need to adjust your FW. Check YouTube. MKS has a series of videos, and one is about UART mode.

themacboy, 2021-06-23:
Linear advance needs Uart mode? Why? There is any reference to that
sentence in Marlin docs?

RFSTER, 2021-06-23:
Sorry I forgot to mention that applies to TMC2208 (which I was assuming you had). It may be different for other stepper drivers, but this is a well known fact for TMC2208. I have a Robin Nano V1.2 with TMC2208. Linear advance made no difference just as you described. I did the modification here: https://youtu.be/7ShcFKXrVHo  ; and then linear advance worked as it is supposed to.

themacboy, 2021-06-24:
> Sorry I forgot to mention that applies to TMC2208 (which I was assuming you had). It may be different for other stepper drivers, but this is a well known fact for TMC2208. I have a Robin Nano V1.2 [...]

Please read my question. Thx

RFSTER, 2021-06-24:
As far as I can tell, your question is" "why "and if "there is any reference in Marlin documentation."
I don't know why, I have not seen reference in Marlin docs, but I just know it is the way it is with TMC2208. If I recall, you also have to run in SpreadCycle mode (which is why you need UART).
This seems to be due to some sort of bug with TMC2208 and Marlin documentation is not going to include every bug for every hardware.
If you spend some time to google "TMC2208 linear advance", you will find a lot of people having this issue; and a lot of people having success by running UART/SpreadCycle mode.
https://github.com/MarlinFirmware/Marlin/issues/11825

Either way, I'm just telling you what works for me with a MKS Robin Nano V1.2 and TMC2208.