makerbase-mks/MKS-Robin-Nano-V1.X #42: Problem with original firmware

grau0, 2020-07-06:
Hello everyone, I just got the nano v1.2 + TFT 35 + TMC2209 MKS V2 + WIFI, and I have a problem with allí original firmware.
The motors only rotate in one direction, even though I try through the LCD it does not rotate (if reversing the rotation in the LCD rotates the other way, but not both ways).
If I load marlin, it works fine, but the icons on the LCD come out without text.
Can somebody help me?
Thank you.

Engi23, 2020-07-07:
It looks like you haven't completed the move home. Immediately after switching on, not all movements work. Make a move home and the movement will work both ways.

makerbase-mks, 2020-07-07:
Can you provide vidoe to me about your board runing? for example, move motoer and so on.