makerbase-mks/MKS-Robin-Nano-V1.X #44: Motor shutdown does not work immediately

Engi23, 2020-07-07:
Mks_Nano35_V2.0.3.1 
1. I go to the home movement menu and start the home movement. Sometimes you need to interrupt the movement, for example, if the switch is far. But pressing “Turn off the motors” does not turn off the motors immediately, but only when the limiter switch is pressed. Sometimes the limit switch cannot be pressed due to a mechanical problem, it is usually one of the first starts and I am ready for it, I click Turn off the motors, but the motor continues to work spoils the belt, etc.
2. Going home during the printing process is also unsafe. Sometimes the limit switch cannot be pressed. The printer is unreasonably trying to move the axis for a long time, I expect that the movement home should not exceed # Travel limits after homing (units are in mm)
((X_MAX_POS)-(X_MIN_POS))*1,2 (20% запаса)

makerbase-mks, 2020-07-08:
For your first question, I think Turn off motor when you moving motor is can not, because firmware this function is need motor enter stop status, so, we need wait motor stop and turn off motor(disable drive)

Engi23, 2020-07-09:
Do not understand the second answer. What distance will the motor travel if the limit switch is not pressed?