makerbase-mks/MKS-Robin-Nano-V1.X #48: MKS PWC does not work when overheated

Engi23, 2020-07-17:
1. I have MKS PWC and I expected that in an emergency (significant overheating) the board would disconnect itself from the power supply, but this did not happen. Heaters are controlled by transistors. It is well known that transistors often turn on during combustion. With settings.> HEATER_0_MAXTEMP 275 
the temperature went up to 350 degrees, why didn't the board disconnect from the network ?! It is obvious. This is generally the main point of payment - to protect the house from fire!

makerbase-mks, 2020-08-11:
Hi! Generally speaking, the thermal protection function of marlin can avoid the problem you mentioned.

mks-viva, 2020-10-22:
@Engi23 Has this problem been solved? I will temporarily close it.

closed 2020-10-22 by mks-viva