makerbase-mks/MKS-Robin-Nano-V1.X #49: Temperature control does not work when I upload files via wifi

Engi23, 2020-07-17:
I turned on the printer and started preheating the M140 S50; M104 S200;
I went to the computer and started sending the wifi via wifi
It's obvious that
1.the temperature of the nozzle did not have time to rise above 100 degrees and the PWM output was about 100%
2. During download, the last state of the PWM or bang-bang output is saved
This time I downloaded a very large file and it took 5 minutes or more to download.
When I approached the printer, smoke came out of the nozzle and the temperature of the nozzle was 350 degrees, with
> HEATER_0_MAXTEMP 275

I often did this, but before that i was just getting error 2 or 6 or 7. But 2 events happened at the same time, I loaded a large file and the nozzle temperature did not have time to stabilize.
Please make sure that the temperature control works during Wifi boot!

M-Coop, 2021-04-30:
Ah I was wondering what the smell of melting plastic on my Flsun Q5 was! Luckily I don't think it did any damage. I assume it's a version of this firmware. Happened whilst I uploaded a file using the wifi. I'll be careful not to preheat

delaosa, 2021-05-06:
Same here with official FLSUN firmware, (V2.5 XYZ 2208 Extruder 4988 firmware), which version are you using?