makerbase-mks/MKS-Robin-Nano-V1.X #55: Can we upgrade MKS-Robin-Nano firmware v2.x.x to MKS-Robin-Nano v1.2 Board

cemiloztas, 2020-07-24:
There is no clear information about it. 
Can we upgrade MKS-Robin-Nano firmware v2.x.x family to MKS-Robin-Nano v1.2 Board. Or we need v2.x board.

And i upload Marlin2.0 to V1.2 board but it does not work. Screen is black.

nictronik99, 2020-07-25:
I'm not sure I understand your question, but I try to answer you: MKS Robin Nano firmware is only for Robin nano v1.1 or 1.2 Both v1.x.x and v2.x.x.
On Robin nano v2 runs only  Marlin 2.x.
To load Marlin 2.x on Robin nano v1.2 you must make sure that the names of the board and the environment are set correctly. When you compile the firmware on VSCode you find compiled firmaware robin_nano35.bin  in the directory .pio / build / mks_robin_nano35 you don't have to use the one in the firmware folder, from that folder you have to take only the other two elements present, not the .bin

thierryzoller, 2020-07-25:
To run marlin 2.0 on Nano Robin 1.1 & 1.2 look here:
https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano

wbucket, 2020-09-26:
I just bought a new v1.2 broad and it came pre-installed with v2 firmware. Hope this means u can flash v2 firmware onto v1.2 broad.