makerbase-mks/MKS-Robin-Nano-V1.X #57: Hotbed temperature issue

WesleyvH, 2020-07-28:
Hello,

I'm building a DIY 3D printer with a MKS Robin Nano v1.2 with 3.5" touchscreen.

I connected everything and it works except for the hotbed, it keeps reading a temperature of 103 degrees Celsius even when it's cold. I tried to heat it up by increasing the temperature to 145 degrees Celsius which is max, and it heats up to about 40 degrees Celsius and then gives me Err6 on screen and stops heating.

Is there a way to fix this? Can i change thermistor type or thermostat type of the hotbed in the firmware?

makerbase-mks, 2020-08-11:
The maximum temperature that the hot bed can reach depends on its power and has nothing to do with the motherboard. You should replace the hot bed with a higher power instead of measuring the temperature.

mks-viva, 2020-11-16:
@WesleyvH 
I will close this question temporarily.

closed 2020-11-16 by mks-viva

Mrjavaci, 2021-10-04:
I am having the same problem.
How do I fix this problem?

WesleyvH, 2021-10-04:
Check your wiring, i accidently switched up 2 wires of the hotbed. I found out, switched them the way they should and that fixed the problem.

Mrjavaci, 2021-10-06:
I solved problem with reset machine firmware.