makerbase-mks/MKS-Robin-Nano-V1.X #61: Hardware: MKS robin nano uses 5v for io voltage supply

Teslafly, 2020-08-21:
Since the micro controller is 3.3v, the io voltage should be 3.3v. this can cause intermittent issues with the tmc2209 driver if you are sending it 3.3v logic and it is using 5v for determining if the signal is high or low. 

at the vary least the logic voltage supply for the stepper motors should be selectable if not permanently changed to 3.3v.

Figured this out after connecting the uart pins on the tmc2209 to the wifi uart and had problems communicating with more than 2 drivers. the uart high voltage would sag too low.

Teslafly, 2020-08-21:
this is on the mks robin nano hw V1.2

mks-viva, 2020-10-23:
@Teslafly Thank you for your feedback, we will consider it comprehensively.

mks-viva, 2020-12-01:
Hi! I will close it temporarily.

closed 2020-12-01 by mks-viva