makerbase-mks/MKS-Robin-Nano-V1.X #68: Missing advanced settings after firmware update

rodrigowebjump, 2020-09-18:
Hi, 

After updating the firmware, the machine settings (step, invert, ...) in the setting menu was missing.
I already tried to updated the firmware using both Marlin, MKS 1.x and 2.x and the menu is never shown.

What I need to do to see this menu again.
In settings I have: Wifi, Fan, about, Filament, continue, language.

There is a blank space between continue and language that was the space reserved for this missing menu.

Thanks

rodrigowebjump, 2020-09-20:
Hi I just discovered that this is enabled by the following line on robin_nano35_cfg.txt.

>MACHINE_PARAMETER		1

How can I enable this menu in Marlin?

mks-viva, 2020-11-07:
If define MACHINE_PARAMETER  1 DIY moudle
If define MACHINE_PARAMETER 0  parameter cann't be change