makerbase-mks/MKS-Robin-Nano-V1.X #70: Nano V1.2 on Delta, is it possible?

pistachin2, 2020-09-22:
Hhello everyone, I have a Nano V1.2 mounted on a CoreXY running like a shot.
I have built a homemade Delta, and wanted to know if anyone has this plate mounted on a Delta.
I have my doubts with the endstops, since the delta use the max and this board does not have them implemented, although MKS indicates that it is valid for deltas.
Does anyone have this board mounted on a delta printer?
Any connection suggestions?
Thank you very much.

makerbase-mks, 2020-09-23:
If use delta, so your X,Y,Z max connect to X Y Z+ endstop

pistachin2, 2020-10-05:
Hello,
So if I use marlin, the configuration would look like this:
_#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
// # define USE_ZMIN_PLUG // a Z probe
#define USE_XMAX_PLUG
// # define USE_YMAX_PLUG
// # define USE_ZMAX_PLUG_
It is correct or I am wrong.
Thank you very much.

mks-viva, 2020-10-22:
You can enable all

mks-viva, 2021-01-18:
Hi! I will close this issues temporarily.

closed 2021-01-18 by mks-viva