makerbase-mks/MKS-Robin-Nano-V1.X #71: Pause to change color M600

ivograca, 2020-09-24:
M600 dosnt work on any versions of firmware

mks-viva, 2020-10-22:
Hi! If you use M600 function, so you need enable it.

ivograca, 2020-10-22:
and how to activate the m600 in the MKS firmware

mks-viva, 2020-10-22:
Which motherboard and screen do you use?

ivograca, 2020-10-22:
mks robin nano 1.2 and 3.5 screen in a two trees bluer

mks-viva, 2020-10-22:
Which firmware and version do you use?

ivograca, 2020-10-22:
https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/tree/master/Nano%20firmware%20v2.x.x
the version is the 2.0.3.4

mks-viva, 2020-11-07:
You need setting it:
1. In robin_nano35_cfg.txt file enable 
```
>MACHINE_PARAMETER		1			#Online parameter(0: disable, 1: enable)
```
2. MKS Robin TFT35: Setting->Machinesettings->Machine->Endstop type->Filament sensor and Open it
3. Your need use fliament sensor connect to MT-DET1 on the motherboard
More information refer to: https://www.youtube.com/watch?v=eubfvnOriuk

mks-viva, 2020-12-01:
Has this issues be resolved, if there is no response, I will temporarily close it.

closed 2020-12-01 by mks-viva