makerbase-mks/MKS-Robin-Nano-V1.X #78: how to use UART mode with TMC2225

system3rd, 2020-11-13:
Hello

I have Robin nano V1.2 with TMC2225 and I don't know how to use UART mode.

as MKS youtube instruction, https://youtu.be/7ShcFKXrVHo?t=34  it should be soldered and wired to wifi socket.

but regarding to tmc2225 datasheet, https://wiki.fysetc.com/TMC2225/  tmc2225 don't need to soldering.

please let me know how to use uart with tmc2225.

Thanks~!

mks-viva, 2020-11-13:
Hi! It is easy. You only need settings like follow:
1. Acording to MKS youtube setting X Y Z E wiring
2. Firmware setting and add pin define:
  - Setting drive type is: TMC2208（）
  - Add pin define
  ```
  #if HAS_TMC_UART
  /**
   * TMC2208/TMC2209 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

  //
  // Software serial
  //

  #define X_SERIAL_TX_PIN                   PA10
  #define X_SERIAL_RX_PIN                   PA10

  #define Y_SERIAL_TX_PIN                   PA9
  #define Y_SERIAL_RX_PIN                   PA9

  #define Z_SERIAL_TX_PIN                   PC7
  #define Z_SERIAL_RX_PIN                   PC7

  #define E0_SERIAL_TX_PIN                  PC13
  #define E0_SERIAL_RX_PIN                  PC13

  // Reduce baud rate to improve software serial reliability
  #define TMC_BAUD_RATE                    19200
  #endif // TMC2208 || TMC2209
  ```

system3rd, 2020-11-13:
> Hi! It is easy. You only need settings like follow:
>
> 1. Acording to MKS youtube setting X Y Z E wiring
> 2. Firmware setting and add pin define:
>
> * Setting drive type is: TMC2208（）
> * Add pin define [...]

Thanks for your reply!!
but tmc 2225 datasheet has different board layout with tmc2208. (R8 and UART -> TX and RX)
do I have to soldering TX and RX on tmc2225? and wiring to wifi socket as mks youtube?

mks-viva, 2020-11-13:
You can connect to RX and test it. Tx need not connect.
[image]

system3rd, 2020-11-13:
> You can connect to RX and test it. Tx need not connect.
> [image]

OK. soldering RX and wiring to wifi socket!! I will try this weekend. Thanks!!

mks-viva, 2020-11-16:
It is will working?

mks-viva, 2020-12-01:
Has this issues be resolved, if there is no response, I will temporarily close it.

closed 2020-12-01 by mks-viva

adamo000, 2022-11-26:
@mks-viva, how about Robin Nano 1.3-s? Is it possible on this version? 
The mainboard has 4x soldered TMC2225 with 1x pluggable driver.

Serox13, 2023-10-02:
> @mks-viva, how about Robin Nano 1.3-s? Is it possible on this version? The mainboard has 4x soldered TMC2225 with 1x pluggable driver.

you just use the TMC 2208 uart configs in klipper and it should work. TMC2225 and 2208 get used on big tree tech boards and they just use TMC2208 uart configs