makerbase-mks/MKS-Robin-Nano-V1.X #80: MARLIN 020007 does not compile on NANo v1.2

Frederic-SABU, 2020-11-25:
Trying to upgrade to MARLIN 020007 on a SAPPHIRE PRO but will not compile on platform.io. MARLIN 020000 works fine but I wanted to get the MARLIN color UI.
Can you help ?

mks-viva, 2020-11-25:
Can you provide your firmware to me and i will build it and test it?

Frederic-SABU, 2020-11-25:
Thks but I have trashed it in the meantime since I have managed to 
compile a MARLIN 2.00.6 adapting it from a FLSSUN Q5 (NANO v1.2 + TFT 2.4).

I wanted to be able to use the MARLIN Color UI with a touch screen but 
in a larger TFT (3.5).

Cheers,
Frd

mks-viva, 2020-11-25:
You use MKS Robin Nano V1.2+MKS Robin TFT35?

Frederic-SABU, 2020-11-25:
No I was referring to a Nano v1.2 with a flat cable 3.5 TFT (OEM on a 
Sapphire pro).

Works great with Touch Screen Marlin (2.0.6) and Color UI.

Frd

You use MKS Robin Nano V1.2+MKS Robin TFT35?

mks-viva, 2020-11-25:
You want to use Nano v1.2 with a flat cable 3.5 TFT run the latest version marlin firmware color UI?

Frederic-SABU, 2020-11-25:
Yes but I am already running version 2.0.6 and I gave a try to 2.0.7 
and got "bool" and "random" errors when compiling on Platform io.

Frd

Le 25/11/2020 11:46, mks-viva a écrit :
>
> You want to use Nano v1.2 with a flat cable 3.5 TFT run the latest 
> version marlin firmware color UI?
>
>

mks-viva, 2020-11-25:
Ok! I will try to test it.

mks-viva, 2020-11-25:
You need set it
In configuration.h file: 
Set board: #define MOTHERBOARD BOARD_MKS_ROBIN_NANO
Enable #define MKS_ROBIN_TFT35 and #define TFT_COLOR_UI
Enable #define TOUCH_SCREEN

In platformio.ini file: set default_envs = mks_robin_nano35

[image]
[Robin_nano35_test_bin.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/files/5596282/Robin_nano35_test_bin.zip)
[Marlin-bugfix-2.0.x.zip](https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/files/5596287/Marlin-bugfix-2.0.x.zip)

Frederic-SABU, 2020-11-25:
> image

No luck, still the same errors, MARLIN from your link.

C:\users\msabu\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish_types.h:65:9: 
error: unknown type name 'bool'

and

C:\users\msabu\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/ext_interrupts.h:90:6: 
error: conflicting types for 'attachInterrupt'

Frd

mks-viva, 2020-12-01:
I think it’s a problem with your compilation environment and has nothing to do with firmware.

Frederic-SABU, 2020-12-01:
Could be.
I am running a Marlin v2.0.6. which is compiling just fine for the Nano 
v1.2 and let me use the MARLIN Color UI and mostly the touch screen so I 
am happy.

Works good for both the TFT 2.4 and 3.5.

Thks for your support anyway.
Rgds,
Frd

mks-viva, 2020-12-02:
Haha! So, I will close this issues.

closed 2020-12-02 by mks-viva

Frederic-SABU, 2021-01-11:
Hello,

PLATFORM IO must have updated something on their board definition since 
ROBIN E3 does not exist anymore and it stopped compiling on my thre E3D 
boards with error :

Marlin\src\HAL\HAL_STM32F1\dogm\../../../inc/../HAL/./HAL_STM32F1/fastio.h:30:10: 
fatal error: libmaple/gpio.h: No such file or directory

Can you help ?

Rgds,
Frd

Frederic-SABU, 2021-01-11:
Pls disregard, forgot to check your GIT.

Rgds,
Frd

Le 11/01/2021 10:30, MSABU a écrit :
> Hello,
>
> PLATFORM IO must have updated something on their board definition
> since ROBIN E3 does not exist anymore and it stopped compiling on my
> thre E3D boards with error : [...]

mks-viva, 2021-01-12:
@Frederic-SABU You can try to use the latest version marlin test it.

Frederic-SABU, 2021-01-12:
This is what I did in the end from your Git and it works.

Rgds,
Frd

Frederic-SABU, 2021-02-17:
I seem to be unable to find a valid Marlin for the NANO v2.0 on your 
github. I need it for a CORE XY machine.

Where can I get it ?

Rgds,
Frd