makerbase-mks/MKS-Robin-Nano-V1.X #81: fillament sensor and step motor

misterry08, 2020-11-25:
2 minutes after the resin identification sensor has run out, all stepper motors are put to rest. It causes a step deviation problem because when changing the resin it moves the X and Y axes because the stepper motor is idle. Can you show me how to keep the stepper motor on to keep the coordinates when out of fillament.

mks-viva, 2020-11-26:
Hi! Which firmware are you using?

misterry08, 2020-11-26:
i use Mks_Nano35_V2.0.3.2

mks-viva, 2020-11-26:
In Configuration_adv.h file
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
[image]

misterry08, 2020-11-26:
I found the place you pointed out but there was another problem :((
[image]
which one should I choose now
[image]
I don't know what to write in this place to stop errors

mks-viva, 2020-11-27:
You need enable a LCD type in configuration.h file, for xexample, if you use MKS Robin TFT35, so you need enable "TFT_LVGL_UI_FSMC"

misterry08, 2020-12-16:
Can you help me fix it?
for me it's really hard.
I use this version so can you edit it and send it to me?
https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/blob/master/Nano%20firmware%20v2.x.x/History%20Version/Mks_Nano35_V2.0.3.2%20Release%20file.rar