makerbase-mks/MKS-Robin-Nano-V1.X #82: other screen by UART

dacentichka, 2020-11-30:
good day, I have a Robin nano v1.2 with the screen, I was able to connect to it the screen Ewart third party, as it failed to do the firmware with a graphical interface and drivers тмс2209 by Ewart and Sensorless homing and Wi-Fi, the chip on the Board does not have enough memory, and now I have no limit switches, wifi is working fine but the problem is that the Board does not start unless I unplug native tft35, please tell me what settings I need to zakommentirovat in the Marlin 2.xxx to get everything working without native screen?

mks-viva, 2020-12-01:
Hi! I don’t seem to know what you mean, can you describe them precisely?

dacentichka, 2020-12-01:
Yes, of course

dacentichka, 2020-12-01:
now I'll take pictures and attach them

mks-viva, 2020-12-01:
Ok! Waiting your pictures...

dacentichka, 2020-12-01:
[image]

this is the screen I want to put on

dacentichka, 2020-12-01:
[image]

this is a screen that I can't disable in the firmware, nothing works without it

mks-viva, 2020-12-02:
Yes, if you need to use UART, then you will not be able to use the original firmware, you need to use the new marlin and support uart mode

dacentichka, 2020-12-02:
the problem is that I built a new Marlin from github Marlin, and the Board without its screen does not work (((tell me what you need to disable in the firmware so that it does not poll the screen when loading??

mks-viva, 2020-12-02:
Haha! You can refer to MKS WIKI：

dacentichka, 2020-12-02:
did you give me the settings for this screen?
[image]

mks-viva, 2020-12-03:
Hi! There is a difference between MKS TFT35 and MKS Robin TFT35. MKS Robin TFT35 is just a display screen. MKS TFT35 has its own CPU that can work independently and communicate with the motherboard through the serial port.

dacentichka, 2020-12-03:
so how do I make Robin nano 1.2 work without my own screen??? I tried all the options from the table above and when the screen is turned off, the Board stops starting, it does not flash without the screen, and if it is controlled from the screen via UART, it does not respond(((

mks-viva, 2021-01-06:
You need to connect the screen to see the firmware update process, otherwise, it is difficult for you to judge what your motherboard is doing at this moment.

dacentichka, 2021-01-09:
good afternoon, you know your board very poorly, I was helped to deal with my problem by another person on github, he knows about your board garazdo more, below I will describe step by step how to get rid of this useless screen and free up space on the microcontroller. The first step in the firmware is to disable these parameters #define ADVANCED_PAUSE_FEATURE
#define FILAMENT_RUNOUT_SENSOR
#define LCD_BED_LEVELING
#define TOUCH_SCREEN
#define TFT_COLOR_UI
#define MKS_ROBIN_TFT35

After building the firmware file Robin_nano35.rename bin to Robin_nano.bin and put it on the memory card. Disconnect the screen from the board and turn it on. To check that everything has started, you can connect via USB, and after restarting the board flashes the LED of the mosfet of the fan blowing the model, you can also look at it, and then on the Wi-Fi pad on the px and tx connect any display on the uart. This works, but if you then flash the board without a native display 5 times in a row, it goes into protection, then you need to connect the native display back and flash the Robin_nano35 file.bin, and then rename again and flash with the display disabled. I tried to describe in great detail all the subtleties of this process if you have any questions, please contact me.

mks-viva, 2021-01-09:
I am sorry, that we did not consider this way of use. We agree that users only use MKS Robin TFT35.

closed 2021-08-09 by mks-viva