makerbase-mks/MKS-Robin-Nano-V1.X #83: Can not update firmware

alikosif, 2020-12-19:
Hi. I have a MKS Robin Nano v1.2. I have updated its firmware to V2.0.3.2 three months ago and I stopped wotking on it at that time. Now today ı want to activate my 3d printer, it can not read the micro SD card. I tried many different formats (ntfs, fat32, exfat). Why I want to update firmware: endstop signal must be inverted.

Can anyone help me?
Thank you.

mks-viva, 2020-12-21:
You Tf card need FAT32

alikosif, 2020-12-21:
I tried Fat 32 but it did not work. Card is 16 gb. not only firmware file, robin nano can not read the tf card including g code. is there different solution?

mks-viva, 2020-12-21:
Can you try another card to test it

alikosif, 2020-12-21:
I tried with a different old 8 gb Fat32 formatted tf card and I can update firmware. I'm sorry to discuss a simple problem here. Thank you.

mks-viva, 2020-12-22:
Ha! So, i will close this issues temporarily

closed 2020-12-22 by mks-viva

pkozlenko, 2021-03-05:
I would like to throw my name in on this issue also.
I currently have the original firmware (never been updated - V2.0.1). I am using an 8GB FAT32 formatted micro SD card. Compiled 'Successfully' using VS Code. Copied 'assets' folder and Robin_nano_v3.bin file created.
I put the SD card into my MKS Robin Nano v1.2 board - and the board boots with no indication that it recognised the SD card whatsoever.
Now - I am not sure whether V3 firmware is too agressive and I should stick to v2 generation. (???)
Any comments?
I am going to try a V2 firmware update now to see if this 'IS' the issue or not.

pkozlenko, 2021-03-05:
Downloaded (MKS-Robin-Nano-V1.X/Nano firmware v2.x.x/Mks_Nano35_V2.0.3.5 Release file.rar) - extracted the file and copied the bin file (Robin_nano35.bin) to the same SD card - message indicated: Updating TFT. Completed. When I select the 'About' on the display it still indicates V2.0.1 for the firmware.
One good point - the SD card was readable.