makerbase-mks/MKS-Robin-Nano-V1.X #92: "#error \"Unknown MOTHERBOARD value set in Configuration.h\"", 	 737,  6,

Lkettlewell, 2021-02-10:
Hello,

I am losing my mind! I have been fighting with, and sadly losing in my battle trying to revive my Ender 3 V1 that hasn't printed since March, poor quality filament damaged my hotend so I changed it for a Triangle Labs V6 Clone, added a Bondtech direct drive extruder, a Triangle Labs NEMA 17 pancake stepper driver, dual Z axis stepper, and paired it all with an MKS Robin Nano V1.2 with the TFT35 (not the cool new SPI version but the ribbon cable version controlling TMC2209 drivers. I also have a MKS BLTouch clone that I need to print a bracket for before installing, as well as I plan to use a Z-MAX sensor, again need to print a mount... I am very much looking forward to having this function again!

I wanted to update the firmware on it with the MKS V2.0.3.5, or whatever the current version is, but found it to be too flakey: not retaining settings, constantly dropping the connection to HE TC giving me an "Err 2" message that would only go away if I reflashed the firmware, it was getting tedious! I have yet to test the stepper motor for the extruder because of the "Err 2" message, I think?! I have checked all of my wiring connections and have everything corrected properly, at least I can get the resistance of every sensor and coil... After x # of days being frustrated, I decided to say "F*** it!" and "I'll just put Marlin on it then!" 

This has also increased my frustration as I had to figure out Visual Studio Code after realising that with Arduino IDE I could not compile the proper .bin as the Nano MB is a 32-bit job, onto platformIO! I have gone through all of the configuration / advanced config files, making the appropriate changes, knowing I will have some tweaking to do after I get through basic operation and need to activate my extra components (BLTouch, Z-MAX, some other shiny bits). I was constantly getting 2 error messages when I would try to compile the workgroup, and I eventually managed to smash my way through the first one; the cleared fault just required redefining some pin names / allocations for the dual Z axis from the dual extruder (who knows maybe I will need to change things around if I ever get to building a cabinet and dry-box, I may need a relaying stepper for the filament?). However, the second fault code I cannot get to clear no matter what I have tried, which is the subject of this post. Now, I have pulled the Motherboard value straight from the .\core\boards.h file via copy / paste and it gives the same error. 

The interesting part is that previously, I also received a "code error for missing binary value" before the 4013 value; the error showed up in the terminal box if I scrolled through the event log. That error code stopped showing up, but that may be due to the compiler failing sooner in its operation as those would happen in the ~15 second runs, while it is currently making ~ 1 second runs before failure? 

Another interesting thing is that while the board is listed as supported, when I tried to use the Marlin Building tool, I also got the same error message and could not get past the first screen. 

I have included my config files [lkettlewell-Marlin](https://drive.google.com/file/d/149rugBH3ZAjK91kBWm_tsFvlq3w5Cfce/view?usp=sharing) and see if there is something that sticks out about that line or some other reason for it not compiling that would be much appreciated. 

Thanks

organizedkaos, 2021-02-21:
I’m posting from my mobile so I can’t see your files.

I can say that I have been able to compile Marlin2.0.7 with no errors onto my MKS Robin Nano v1.2 with TFT35, tmc2209 drivers in uart mode and sensorless homing on xy, and bltouch.

I compiled marlin with a few changes though which was mainly for the uarts, sensorless homing and I changed from the lvgl gui to the marlin built in tft gui.

It can be done.

Dom749s, 2021-03-02:
Hello,

Modify
```
#ifndef MOTHERBOARD
  #define BOARD_MKS_ROBIN_NANO       4013  // MKS Robin Nano (STM32F103VET6)
#endif
```
to
#ifndef MOTHERBOARD
  #define BOARD_MKS_ROBIN_NANO    // MKS Robin Nano (STM32F103VET6)
#endif

closed 2021-08-09 by mks-viva