makerbase-mks/MKS-Robin-Nano-V1.X #97: TMC2208 Heating

acca84, 2021-03-11:
Hello
I'm wiring A Mks robin nano v1.2 on Tronxy D01 printer
I use TMC2208 for x, y, z and extruder axis
all axis move correctly but one of the driver becomes very hot as soon as the printer is turned on. Other drivers don't have this problem and have the same Current set in marlin firmware

am i lossing something ?

madurani, 2021-03-11:
I had same problem, but current I set up manually via potentiometer on driver TMC2209. My steppers motors have current 0.9A and on driver I set up current 0.6A. After this change driver and motor aren't more hot.

acca84, 2021-03-12:
Ok, thanks my motors have 1,68A rated current, the default voltage on TMC2208 is 1,1v it seams ok if i apply 1,68/1,41
Am i right ?

Is that supposed to apply the correct current if i declare TMC2208 (and not TMC2208 standalone) in Marlin firmware and don't wire UART to Wifi connectors ?

madurani, 2021-03-12:
Should be OK, but very good support for MKS robin nano v1.2 you can find here:https://www.youtube.com/watch?v=d52u0Mss7fI&lc=UgxGzGn41DB5plWRb1F4AaABAg.9KGjfEzHp6S9KKsx5oR9qi
Under video in chat, you can find with my name solution for same issue with a current and hot driver and motor.
In case, that you have some question you can write there and Makerbase Team will response on your question.