makerbase-mks/MKS-Robin-Nano-V1.X #103: I can't upgrade the firmware MKS Robin Nano V1.2

Neo70777, 2021-04-04:
Hi
I can't upgrade the MKS Robin Nano V1.2 from TF sd card slot (micro-Sd) in mainboard because it can't read it. I Try with multiple Sdcards and the issue persist.

How can it Upgrade the mainboard?.

Thank you

mks-viva, 2021-04-14:
1、Copy Robin_nano35.bin to TF card
2、Insert TF card to board and repower on

Neo70777, 2021-04-14:
Thank you for your support. 

I found the issue now and test it.

I format to Fat 16 and fat 32 the micro sd  from an SD adapter. If format from it the mainboard it don't detect the Micro Sd. I format the micro Sd from smartphone and now detect the Micro Sd card and i upgrade the Mainboard.

Thank so much for your support and attention.

closed 2021-04-14 by Neo70777