makerbase-mks/MKS-Robin-Nano-V1.X #105: Design UI

misterry08, 2021-04-21:
Can you give me the design of UI Version 2.0.6. I need to redesign my style.
Thanks!

misterry08, 2021-04-21:
I use Adobe Illustrator

gensoloone, 2021-04-26:
You take the LogoBuilde program. You translate the icon in the mks_pic folder from the format .bin to image format. You edit it. You convert the edited image back to the bin format. And so with all the files in the mks_pic folder.
You can also use the service for configuring firmware for Robin series cards https://baizhongyun.cn/home/mkstoolview (but there are not all the icons that are in the firmware). I changed the design in both ways. First through the site, then, those icons that can not be changed through the site, edited through LogoBuilder.
----------------------------------------------------------------------------------------------------------------------------------------------
Берешь программу LogoBuilde. Переводишь иконку в папке mks_pic из формата .bin в формат изображения. Редактируешь. Переводишь отредактированное изображение обратно в формат bin. И так со всеми файлами в папке mks_pic.
Также можно воспользоваться сервисом по конфигурации прошивок для плат серии Robin https://baizhongyun.cn/home/mkstoolview (но там не все иконки которые есть в прошивке). Я изменял дизайн обоими способами. Сперва через сайт, затем, те иконки которые нельзя изменить через сайт, редактировал через LogoBuilder.

misterry08, 2021-04-27:
tôi đã làm được điều bạn làm rồi.. Cảm ơn. Thật ra tôi là một nhà thiết kế. Nhưng tôi chỉ cần đổi màu của các biểu tượng nên nếu vẽ lại thì khá lâu. Nên tôi mới hỏi xin thiết kế gốc dạng .Ai hoặc .PSD để đổi cho nhanh