makerbase-mks/MKS-Robin-Nano-V1.X #108: Babystepping Z Offset

WashingtonJunior, 2021-07-12:
Version 2.0.3.6 is not showing babystepping Z offset value (although it moves the axis). Previous version, 2.0.3.5, shows it normally.

Thanks!

steve007iii, 2021-10-21:
It is also difficult to define when an automatic leveling is performed.
babystepping Z offset  values shown on the screen do not work

Smarthelpers, 2022-01-08:
Could anyone find a solution for this problem?

WashingtonJunior, 2022-01-09:
> Could anyone find a solution for this problem?

I've been using version 2.0.3.5, until they release a new version fixing that. This is the closed-source version, so only MKS developers can fix it.

Smarthelpers, 2022-01-09:
Thank you for this information - you are right, the 2.0.3.5 is much better then the 3.6. I made my own firmware with marlin, but unfortunately I could not find a way for getting the z probe wizard included. Its a bit hard to get informations about it and mks seems to Focusing on new products only.

WashingtonJunior, 2022-01-09:
> Thank you for this information - you are right, the 2.0.3.5 is much better then the 3.6. I made my own firmware with marlin, but unfortunately I could not find a way for getting the z probe wizard included. Its a bit hard to get informations about it and mks seems to Focusing on new products only.

Which printer do you own? I've tried their open source Marlin version with Two Trees Bluer and BL-Touch and babystepping worked as usual (but I'm having issues with skipping steps on Y axis after first layer, so I switched back to this 2.0.3.5).

Smarthelpers, 2022-01-09:
I developed my own printer, but use the MKS Robin Nano V1.2 Mainboard - all runs fine, but I am missing the Z-Offset-Wizard for changing the Z-Offset comfortable after automatic bed leveling. In the Elegoo Neptune 2 I have this wizard and this printer using a similar board with the same TFT35.