makerbase-mks/MKS-Robin-Nano-V1.X #112: How to update firmware *without* TFT screen attached?

glabifrons, 2021-08-29:
I'm not planning to use the display, but firmware will **_only_** flash if the TFT35 display is attached.

I found the recommendation to rename the firmware from Robin_nano35.bin to Robin_nano.bin in two other discussions: #64 & #82 
Unfortunately, that is not working on my board.

I can repeatedly flash with success if both the TFT LCD is attached **_and_** the file is named Robin_nano35.bin.
However, if I disconnect the display, I cannot find a filename the firmware will accept for flashing (I've tried Robin_nano35.bin, Robin_nano.bin, and even Robin.bin).

Note that this board **_is_** available new without the display - so how do people update their firmware without the LCD?

katodo, 2021-10-07:
is there any news about this issue? I need to use this board with klipper and without tft, can be possible?

madurani, 2021-10-07:
I use board RobinNanoV2 without display, because equipment EXP-MOT3 with next three drivers/stepper was replaced with display.
Update of firmware is same as with display: only sd-casd with file Robin_nano35.bin insert to card-reader, start board and wait 30 second. 

Only one issue on me side: I have to pull out cable from socket exp1 because motor z4 in socket m3 begin horrible sound/movement generate.

glabifrons, 2021-10-10:
> Thanks! I hope to try this solution next weekend!
> 
> isn't available an open bootloader firmware? I need to use this board with klipper.

You flash Klipper to it by putting the image on a microSD card and powercycling the printer with the card in the Robin Nano.
The bootloader doesn't get replaced by the flashing, as it's in a part of the flash that doesn't get overwritten by this process.

Klipper is why I opened this issue (I can't use the stock LCD with Klipper, so I replaced it with a Pi LCD running KlipperScreen).

Germwalker, 2021-11-23:
I have the same problem, i can't flash anything without the TFT screen detached. This is so anoying, the board is integrated deeply inside the printer and i need to unscrew everything to update the klipper firmware.

Germwalker, 2021-11-23:
I'm tired with this crappy bootloader. I think i'll try to go to SWD with an STLINK to flash the klipper firmware.

Kung-Foo-Kamel, 2022-04-27:
I had no issues flashing my Nano-S V1.3 with Klipper?

Flash file must be called robin_nano35.bin and you will need to compile a newer firmware via klipper as this board runs an STM32F407 MCU and has 32kb boot flash vs 28kb on previous boards.

You must have no display connected, also make sure you have the USB cable disconnected, power up the printer and leave it for a few minutes, power down, remove the card, check the card in your PC, if it has been renamed "robin_nano32.CUR" then flash was successful.

NexGen-3D-Printing, 2022-05-01:
I would only do this if there was an issue with the stock bootloader, there may have been something corrupt with yours?

I have created an updated config for this board: https://github.com/NexGen-3D-Printing/klipper/blob/master/config/generic-mks-robin-nano-s-v1.3.cfg

Germwalker, 2022-05-02:
@NexGen-3D-Printing i was tired to deal with the robin nano stock bootloader as i needed to connect the screen every time i flash klipper and connect a SD card for this only purpose. I had no place in my setup to do so (and i don't have the screen anymore), that's why it's easier for me. The only important point is to compile the bootloader with the good memory offset as the bootloader have been erased.

NexGen-3D-Printing, 2022-05-03:
> @NexGen-3D-Printing i was tired to deal with the robin nano stock bootloader as i needed to connect the screen every time i flash klipper and connect a SD card for this only purpose. I had no place [...]

That's odd, I find my board wont flash "with" the display connected, I had to disconnect it for a successful flash to occur, I have also removed the display and I run an RPI with 4.3 touch screen instead.