makerbase-mks/MKS-Robin-Nano-V1.X #113: Freezing update!

pacocatech, 2021-08-30:
Hello,
I bought a Two Trees Bluer and it came with the MKS Nano S V1.3 board as shown in the image:

[image]

When trying to update the FW, the printer does not turn on anymore and always stays at the same point, as shown:

[image]

what can I do?

Bergateo, 2021-09-11:
same issue but haven't find the solution... do you find a way or have you bought a new one?

pacocatech, 2021-09-11:
nothing so far

mks-viva, 2021-09-13:
@pacocatech 
You can try to this issues and i reply it: https://github.com/makerbase-mks/MKS-Robin-Nano-S-Firmware/issues/3

pacocatech, 2021-09-13:
still with the same problem

mks-viva, 2021-09-16:
May be reupload bootloader to board
https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/tree/master/Bootloader