makerbase-mks/MKS-Robin-Nano-V1.X #114: Robin TFT24/28/32/35/43 screen drivers

aSocialMenace, 2021-09-04:
So i am gonna convert my Neptune 2 to Klipper and it would be nice to use the factory TFT screen with Octoprint as Klipper currently does not support touch screens. I will need to wire up the factory screen (Robin TFT24/28/32/35/43 screen) directly to my pi and install the drivers to get it to work with a plugin like octoscreen. Soooo I am looking for drivers and a pinout of the screen its self. Do any such drivers exist or are there generic drivers that might work with for this? Any help would be greatly appreciated!!

willngton, 2021-09-30:
Hey @aSocialMenace  ,that is a very good idea, however,  drive will need to be written from scratch, for what I saw, all TFT are based on FSMC, I haven't seen any drive or dst for this kind of device, only SPI, 

Been working on try to get TS35 working with PI, (so far got the TFT but not touch)

[image]

robbz23, 2024-03-25:
@willngton I know this is many years later, but I also have a Two Trees SP-3 that I am trying to get Klipper going on the screen.  Do you have any instructions on how to build the firmware to work with MKS Robin TFT35?