makerbase-mks/MKS-Robin-Nano-V1.X #115: Incorrect functionality of UART mode

madurani, 2021-09-04:
I tried to configure on RobinNanoV1.2 UART mode(on axis y) with drivers tmc2208, tmc2209, according "how to" on link https://www.youtube.com/watch?v=7ShcFKXrVHo.
Everything worked on first time, but motor after finishing given motion (even though it works properly) shivers and hisses loudly, however when I switch back to step/dir mode motion of motor is correct(same motor&driver).

Look like that issue is related with kind of stepper motors, because when i use old motors https://www.aliexpress.com/item/32811604070.html?spm=a2g0s.9042311.0.0.27424c4d79bGIL, then problem with hissing sounds occurs, though when I use motor https://www.aliexpress.com/item/32811604070.html?spm=a2g0s.9042311.0.0.27424c4d79bGIL it works correctly(without mentioned sound). 
Is it possible to modify marlin configuration for old kind of motors? 

What was done for UART configuration:
1 three jumpers under driver were removed
2 cable was soldered on pins of driver(uart and R8) and connected to pin of board (PA9)
3 marlin configuration:
Configuration.h >> #define X_DRIVER_TYPE TMC2208 or #define X_DRIVER_TYPE TMC220 //according installed driver
Configuration_adv.h: #define X_CURRENT 636 // max current of motor is 0.9A
Configuration_adv.h: #define X_MICROSTEPS 16
Configuration_adv.h: #define HYBRID_THRESHOLD
Configuration_adv.h: #define X_HYBRID_THRESHOLD 100
pins_MKS_ROBIN_NANO.h #define Y_SERIAL_TX_PIN                   PA9
 pins_MKS_ROBIN_NANO.h #define Y_SERIAL_RX_PIN                  PA9

Driver tmc2208 and tmc2209 behaves the same way 

Here is debug outputs from M122:
```
Connecting...
Printer is now online.
>>> m122
SENDING:M122
		Y	E1
Address		0
Enabled		false	false
Set current	636	636
RMS current	611	611
MAX current	862	862
Run current	19/31	19/31
Hold current	9/31	9/31
CS actual	9/31	9/31
PWM scale
vsense		1=.18	1=.18
stealthChop	false	false
msteps		16	16
interp		true	true
tstep		max	max
PWM thresh.	79	277
[mm/s]		100	30
OT prewarn	false	false
pwm scale sum	11	11
pwm scale auto	0	0
pwm offset auto	36	36
pwm grad auto	14	14
off time	3	3
blank time	24	24
hysteresis
 -end		-1	-1
 -start		1	1
Stallguard thrs	0
uStep count	8	8
DRVSTATUS	Y	E1
sg_result	0
stst
olb		*	*
ola		*	*
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		Y	0x80:09:00:C0
		E1	0x80:09:00:C0
Testing Y connection... OK
Testing E1 connection... OK
```

madurani, 2021-09-04:
After adding all 5 drivers to UART mode + dual z + sensor less on xy axis, all work as expected, without mentioned sound(reason of issue wasn't find)

```
  >>> m122
SENDING:M122
                X       Y       Z       Z2      E
Address         0       0                       0
Enabled         false   false   false   false   false
Set current     800     800     636     800     800
RMS current     795     795     611     795     795
MAX current     1121    1121    862     1121    1121
Run current     25/31   25/31   19/31   25/31   25/31
Hold current    12/31   12/31   9/31    12/31   12/31
CS actual       12/31   12/31   9/31    12/31   12/31
PWM scale
vsense          1=.18   1=.18   1=.18   1=.18   1=.18
stealthChop     true    true    true    true    true
msteps          16      16      16      16      16
interp          true    true    true    true    true
tstep           max     max     max     max     max
PWM thresh.     79      79      658     658     277
[mm/s]          100     100     3       3       30
OT prewarn      false   false   false   false   false
pwm scale sum   26      61      54      31      14
pwm scale auto  0       3       4       1       0
pwm offset auto 64      144     163     76      36
pwm grad auto   17      61      14      14      14
off time        3       3       3       3       3
blank time      24      24      24      24      24
hysteresis
 -end           -1      -1      -1      -1      -1
 -start         1       1       1       1       1
Stallguard thrs 150     150                     0
uStep count     216     24      264     264     8
DRVSTATUS       X       Y       Z       Z2      E
sg_result       78      48                      0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
                X       0xC0:0C:00:00
                Y       0xC0:0C:00:00
                Z       0xC0:09:00:00
                Z2      0xC0:0C:00:00
                E       0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK

```

mks-viva, 2021-09-06:
[image]
This is OK.

madurani, 2021-09-07:
Which sort of memory(flash, eeprom, sdcard) use RobinNanoV1.2 for store of parameters from command M500,  when is set up in marlin?

```
/
// EEPROM
//
#if EITHER(NO_EEPROM_SELECTED, FLASH_EEPROM_EMULATION)
  #define FLASH_EEPROM_EMULATION
  #define EEPROM_PAGE_SIZE     (0x800U) // 2KB
  #define EEPROM_START_ADDRESS (0x8000000UL + (STM32_FLASH_SIZE) * 1024UL - (EEPROM_PAGE_SIZE) * 2UL)
  #define MARLIN_
```
 **Flash memory?**

mks-viva, 2021-09-08:
Use Flash

closed 2021-09-24 by mks-viva