makerbase-mks/MKS-Robin-Nano-V1.X #120: nano-s uart not working for tmc2209

pznamenskii, 2021-10-19:
hi. 
I am trying to install tmc2209 into E1 socket on nano-s board and connect via uart (4th pin on tmc2209 (PDN) as TX to PC14 via 1k resistor and as RX directly to PC15:
```
Marlin/src/pins/stm32f4/pins_MKS_ROBIN_NANO_V1_3_F4.h:
#define E0_SERIAL_TX_PIN                  PC14
#define E0_SERIAL_RX_PIN                  PC15
#define TMC_BAUD_RATE 19200

Marlin/Configuration_adv.h:
#define E0_SLAVE_ADDRESS 0
#define E0_CURRENT      550
#define E0_MICROSTEPS    32
#define STEALTHCHOP_E
#define MONITOR_DRIVER_STATUS
#define TMC_DEBUG

Marlin/Configuration.h:
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V1_3_F4
#define E0_DRIVER_TYPE TMC2209
```
driver is no seen with the below M122 S0 response:
```
09:52:10.380 : Driver registers:
09:52:10.380 : E	0x00:00:00:00	 Bad response!
09:52:10.381 : Testing E connection... Error: All LOW
```

mks-viva, 2021-11-23:
You only need to use an IO to connect to the PDN_UART pin of TMC2209, you can refer to：https://www.youtube.com/watch?v=7ShcFKXrVHo

pznamenskii, 2021-11-23:
> You only need to use an IO to connect to the PDN_UART pin of TMC2209, you can refer to：https://www.youtube.com/watch?v=7ShcFKXrVHo

doing so will likely damage your board: nano-s board has MS3 hardwired with a 0Ohm resistor and until your remove it, UART pin on driver will get voltage as in a non-UART configuration. there is no way to connect UART without soldering out that resistor which is an extremely inconvenient board design.

mks-viva, 2021-11-23:
TMC2209 compatible with 3.3V and 5V

pznamenskii, 2021-11-23:
> TMC2209 compatible with 3.3V and 5V

it is, but UART will not work until you remove this resistor
[image]
by not removing it, you load 3.3 or 5V to the pin on the board you chose to use as UART line, i.e. PC14 - this may damage the board

mks-viva, 2021-11-23:
You need remove it, because it is use to microstep settings for plug driver.