makerbase-mks/MKS-Robin-Nano-V1.X #138: Erratic temp Spikes

Serox13, 2023-10-01:
Hi all,
i have scoured the depths of Google for answers but i need the collective help of Klipper.
I'm using an MKS nano S v1.3 board, it flashed into Klipper no worries did all the usual calibrations except the temps of everything even the mcu would fluctuate to the point of shutting the mcu down. power down then on the next day it wouldn't connect to the MCU at all. Connected an old 4.2.2 board up it ran perfectly connected to the MCU, no temp fluctuations at all, ran a print all was good. 
i got the MKS board working again connected it to the MCU and again temp fluctuations so i PID calibrate and it doesn't fix or stabilise the temps at all. I also turned off the PSU and left the USB cable to power the board and instantly the temps went to normal no erratic fluctuations. These happen with the stock firmware as well.
I have tested all grounds and earths, PSU puts out 24v, no shorts on the board. changed the thermistor even know its not the problem and still happens, i have also made a data-only USB cable and still it happens. Because it levels out after the PSU is turned off does that suggest its a grounding issue with the PSU or MKS board issue? Do i just accept that the board has lots its will to print and RMS it? any help would be great thanks.
If you need logs or what i can provide them when i get back to the main pc.

amsmith525, 2023-11-17:
I too have this issue. Powering from usb only and the readings are flat - turn on the psu and the readings go erratic.

amsmith525, 2024-02-16:
After increasing the value of capacitor C10 from 0.1uf to 10uf, the temp spikes are significantly reduced. 
[image]
[image]